mavlink/mavros #1984: QGroundControl can't connect Mavros

brain0071, 2024-09-04:
### Issue details
I want to use QGroundControl on a laptop to connect to a MAVROS node running on another Jetson NX. The host address running QGroundControl is '192.168.0.3', and the IP address of the Jetson NX is '192.168.0.9'.

My settings in apm.launch are as follows:
```
<arg name="gcs_url" default="tcp-l://192.168.0.9:5760" />
```
However, QGroundControl does not show a connection.
The log information from Mavros is as follows:
```
[ INFO] [1724914089.845271840]: tcp1: Got client, id: 2, address: 192.168.0.3:58962
```

Then, I modified the settings in apm.launch as follows:
```
<arg name="gcs_url" default="tcp-l://192.168.0.3:5760" />
```
The log information from Mavros is as follows:
```
[ INFO] [1724914292.466661824]: FCU URL: /dev/ttyACM0:57600
[ INFO] [1724914292.471926432]: serial0: device: /dev/ttyACM0 @ 57600 bps
[ INFO] [1724914292.473727072]: GCS URL: tcp-l://192.168.0.3:5760
[ INFO] [1724914292.474614976]: tcp1: Bind address: 192.168.0.3:5760
[FATAL] [1724914292.475251264]: GCS: DeviceError:tcp-l:bind: Cannot assign requested address
[ INFO] [1724914292.563975072]: Found a device with ID: 0080003D5E @ port: /dev/ttyUSB0, baudrate: 115200
[ INFO] [1724914292.564094240]: Opening port /dev/ttyUSB0 ...
================================================================================REQUIRED process [mavros-1] has died!
process has finished cleanly
log file: /home/dlmux/.ros/log/671d3d26-65d0-11ef-b7f4-48b02dd8d7a6/mavros-1*.log
Initiating shutdown!
================================================================================
```

How should I modify this setting? Should the address in apm.launch be ```192.168.0.3``` or ```192.168.0.9```? Currently, neither seems to work.

### MAVROS version and platform

Mavros: ?0.18.4?
ROS: Noetic
Ubuntu: 20.04

### Autopilot type and version

[ yes ] ArduPilot

Version: Sub-4.1

vooon, 2024-09-04:
You cannot bind tcp socket to remote address. Simply use `tcp-l://0.0.0.0:5760` for `gcs_url`.

brain0071, 2024-09-05:
> You cannot bind tcp socket to remote address. Simply use `tcp-l://0.0.0.0:5760` for `gcs_url`.

Sorry, I have already tried this setup.But it cannot solve this problem.
```
tcp-l://0.0.0.0:5760` for `gcs_url`.
```

Mavros Log show shows as follows:
```
[ INFO] [1724913239.175047136]: tcp1: Got client, id: 2, address: 192.168.0.3:58580
```

QGoundControl still shows as disconnected.

brain0071, 2024-09-22:
> You cannot bind tcp socket to remote address. Simply use `tcp-l://0.0.0.0:5760` for `gcs_url`.

I found a new question about why QGC and MAVROS can't fight for a connection. It may be due to the FCU not being properly connected.

**diagnostics**

```
rostopic echo /diagnostics 

header: 
  seq: 62
  stamp: 
    secs: 1726978161
    nsecs: 590557832
  frame_id: ''
status: 
  - 
    level: 1
    name: "mavros: FCU connection"
    message: "not connected"
    hardware_id: "/dev/ttyACM0:57600"
    values: 
      - 
        key: "Received packets:"
        value: "0"
      - 
        key: "Dropped packets:"
        value: "0"
      - 
        key: "Buffer overruns:"
        value: "0"
      - 
        key: "Parse errors:"
        value: "0"
      - 
        key: "Rx sequence number:"
        value: "0"
      - 
        key: "Tx sequence number:"
        value: "0"
      - 
        key: "Rx total bytes:"
        value: "0"
      - 
        key: "Tx total bytes:"
        value: "26380"
      - 
        key: "Rx speed:"
        value: "0.000000"
      - 
        key: "Tx speed:"
        value: "460.000000"
  - 
    level: 2
    name: "mavros: GPS"
    message: "No satellites"
    hardware_id: "/dev/ttyACM0:57600"
    values: 
      - 
        key: "Satellites visible"
        value: "0"
      - 
        key: "Fix type"
        value: "0"
      - 
        key: "EPH (m)"
        value: "Unknown"
      - 
        key: "EPV (m)"
        value: "Unknown"
  - 
    level: 1
    name: "mavros: Mount"
    message: "Can not diagnose in this targeting mode"
    hardware_id: "/dev/ttyACM0:57600"
    values: 
      - 
        key: "Mode"
        value: "255"
  - 
    level: 2
    name: "mavros: Heartbeat"
    message: "No events recorded."
    hardware_id: "/dev/ttyACM0:57600"
    values: 
      - 
        key: "Heartbeats since startup"
        value: "0"
      - 
        key: "Frequency (Hz)"
        value: "0.000000"
      - 
        key: "Vehicle type"
        value: "Generic micro air vehicle"
      - 
        key: "Autopilot type"
        value: "Generic autopilot"
      - 
        key: "Mode"
        value: ''
      - 
        key: "System status"
        value: "Uninit"
  - 
    level: 0
    name: "mavros: System"
    message: "Normal"
    hardware_id: "/dev/ttyACM0:57600"
    values: 
      - 
        key: "Sensor present"
        value: "0x00000000"
      - 
        key: "Sensor enabled"
        value: "0x00000000"
      - 
        key: "Sensor health"
        value: "0x00000000"
      - 
        key: "CPU Load (%)"
        value: "0.0"
      - 
        key: "Drop rate (%)"
        value: "0.0"
      - 
        key: "Errors comm"
        value: "0"
      - 
        key: "Errors count #1"
        value: "0"
      - 
        key: "Errors count #2"
        value: "0"
      - 
        key: "Errors count #3"
        value: "0"
      - 
        key: "Errors count #4"
        value: "0"
  - 
    level: 2
    name: "mavros: Battery"
    message: "No data"
    hardware_id: "/dev/ttyACM0:57600"
    values: 
      - 
        key: "Voltage"
        value: "-1.00"
      - 
        key: "Current"
        value: "0.0"
      - 
        key: "Remaining"
        value: "0.0"
  - 
    level: 2
    name: "mavros: Time Sync"
    message: "No events recorded."
    hardware_id: "/dev/ttyACM0:57600"
    values: 
      - 
        key: "Timesyncs since startup"
        value: "0"
      - 
        key: "Frequency (Hz)"
        value: "0.000000"
      - 
        key: "Last RTT (ms)"
        value: "0.000000"
      - 
        key: "Mean RTT (ms)"
        value: "0.000000"
      - 
        key: "Last remote time (s)"
        value: "0.000000000"
      - 
        key: "Estimated time offset (s)"
        value: "0.000000000"
---
```

**check ID**
```
rosrun mavros checkid 
ERROR. I got 0 addresses, but not your target 1:1

---
Received 0 messages, from 0 addresses
sys:comp   list of messages
```

Additionally, I use mavros to send PWM messages to Ardupilot and then the motor executes correctly. So, mavros just cannot accept feedback messages from pixhawk.

vooon, 2024-09-22:
Are you sure that the port and baud rate correct? And your wiring is ok?

brain0071, 2024-09-23:
Yes, I'm sure. baud rate is set: 
```
<arg name="fcu_url" default="/dev/ttyACM0:57600" />
```
I can find the Pixhawk, and it has the appropriate permissions.
```
crw-rw----   1 root  dialout 166,   0 9月  22 10:48 ttyACM0
```
mavros can start normally, and I can even send messages to Pixhawk to control the motors.
```
[ INFO] [1726972517.500111464]: Known MAVLink dialects: common ardupilotmega ASLUAV AVSSUAS all csAirLink cubepilot development icarous matrixpilot paparazzi standard storm32 uAvionix ualberta
[ INFO] [1726972517.500167688]: MAVROS started. MY ID 1.240, TARGET ID 1.1
```
But，it seems to be missing this line of LOG, and when mavros is started, I cannot find the following:
```
[ INFO] [1727001626.144837536]: CON: Got HEARTBEAT, connected. FCU: ArduPilot
[ INFO] [1727001626.146686700]: RC_CHANNELS message detected!
[ INFO] [1727001626.146887262]: IMU: Raw IMU message used.
[ WARN] [1727001626.147318715]: GP: No GPS fix
[ WARN] [1727001626.148618960]: TM: Wrong FCU time.
[ WARN] [1727001627.156981084]: CMD: Unexpected command 520, result 0

[ INFO] [1727001636.151804510]: FCU: ArduSub V4.1.2 (2dd0bb7d)
[ INFO] [1727001636.152267398]: FCU: ChibiOS: 93e6e03d
[ INFO] [1727001636.152434106]: FCU: Pixhawk1 0032002A 3339510F 34383938
[ INFO] [1727001636.152570226]: FCU: RCOut: PWM:1-12
[ INFO] [1727001636.152679348]: FCU: IMU0: fast sampling enabled 8.0kHz/1.0kHz
```

vooon, 2024-09-23:
I asked because Rx == 0 bytes. Have you tried same port/baud with QGC?

brain0071, 2024-10-11:
I have solved this problem and it was caused by restarting. My mavros launch file contains another xsens.launch. This xsens.launch often fails to start, so I use ```ctrl+c``` to terminate the ros node and then execute it again.

When I restarted on the same terminal, the following LOG was no longer displayed, causing the connection to fail.
```
[ INFO] [1727001626.144837536]: CON: Got HEARTBEAT, connected. FCU: ArduPilot
[ INFO] [1727001626.146686700]: RC_CHANNELS message detected!
[ INFO] [1727001626.146887262]: IMU: Raw IMU message used.
[ WARN] [1727001626.147318715]: GP: No GPS fix
[ WARN] [1727001626.148618960]: TM: Wrong FCU time.
[ WARN] [1727001627.156981084]: CMD: Unexpected command 520, result 0

[ INFO] [1727001636.151804510]: FCU: ArduSub V4.1.2 (2dd0bb7d)
[ INFO] [1727001636.152267398]: FCU: ChibiOS: 93e6e03d
[ INFO] [1727001636.152434106]: FCU: Pixhawk1 0032002A 3339510F 34383938
[ INFO] [1727001636.152570226]: FCU: RCOut: PWM:1-12
[ INFO] [1727001636.152679348]: FCU: IMU0: fast sampling enabled 8.0kHz/1.0kHz
```
I solved this problem by separating the two launch files.

closed 2024-10-11 by vooon