mavlink/mavros #1986: catkin build  Report an error “Errors     << mavros_extras:make /home/cwkj/catkin_ws/logs/mavros_extras/build.make.001.log ”

ZHCLN, 2024-09-13:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details

_Please describe the problem, or desired feature_

### MAVROS version and platform

Mavros: 1.19.0
ROS: Noetic 
Ubuntu: 20.04

### step
https://github.com/mavlink/mavros/blob/master/mavros/README.md#source-installation

catkin build make an error!

### error
Errors     << mavros_extras:make /home/cwkj/catkin_ws/logs/mavros_extras/build.make.001.log
/home/cwkj/catkin_ws/src/mavros/mavros_extras/src/plugins/gps_status.cpp: In member function ‘void mavros::extra_plugins::GpsStatusPlugin::handle_gps2_raw(const mavlink_message_t*, mavlink::common::msg::GPS2_RAW&)’:
/home/cwkj/catkin_ws/src/mavros/mavros_extras/src/plugins/gps_status.cpp:107:40: error: ‘struct mavlink::common::msg::GPS2_RAW’ has no member named ‘alt_ellipsoid’
  107 |   ros_msg->alt_ellipsoid     = mav_msg.alt_ellipsoid;
      |                                        ^~~~~~~~~~~~~
/home/cwkj/catkin_ws/src/mavros/mavros_extras/src/plugins/gps_status.cpp:108:40: error: ‘struct mavlink::common::msg::GPS2_RAW’ has no member named ‘h_acc’
  108 |   ros_msg->h_acc             = mav_msg.h_acc;
      |                                        ^~~~~
/home/cwkj/catkin_ws/src/mavros/mavros_extras/src/plugins/gps_status.cpp:109:40: error: ‘struct mavlink::common::msg::GPS2_RAW’ has no member named ‘v_acc’
  109 |   ros_msg->v_acc             = mav_msg.v_acc;
      |                                        ^~~~~
/home/cwkj/catkin_ws/src/mavros/mavros_extras/src/plugins/gps_status.cpp:110:40: error: ‘struct mavlink::common::msg::GPS2_RAW’ has no member named ‘vel_acc’
  110 |   ros_msg->vel_acc           = mav_msg.vel_acc;
      |                                        ^~~~~~~
/home/cwkj/catkin_ws/src/mavros/mavros_extras/src/plugins/gps_status.cpp:111:40: error: ‘struct mavlink::common::msg::GPS2_RAW’ has no member named ‘hdg_acc’
  111 |   ros_msg->hdg_acc           = mav_msg.hdg_acc;
      |                                        ^~~~~~~
make[2]: *** [CMakeFiles/mavros_extras.dir/build.make:232：CMakeFiles/mavros_extras.dir/src/plugins/gps_status.cpp.o] 错误 1
make[2]: *** 正在等待未完成的任务....
make[1]: *** [CMakeFiles/Makefile2:735：CMakeFiles/mavros_extras.dir/all] 错误 2
make: *** [Makefile:141：all] 错误 2
cd /home/cwkj/catkin_ws/build/mavros_extras; catkin build --get-env mavros_extras | catkin env -si  /usr/bin/make --jobserver-auth=3,4; cd -

...............................................................................
Failed     << mavros_extras:make           [ Exited with code 2 ]              
Failed    <<< mavros_extras                [ 30.5 seconds ]                    
Abandoned <<< test_mavros                  [ Unrelated job failed ]            
[build] Summary: 4 of 6 packages succeeded.                                    
[build]   Ignored:   None.                                                     
[build]   Warnings:  None.                                                     
[build]   Abandoned: 1 packages were abandoned.                                
[build]   Failed:    1 packages failed.                                        
[build] Runtime: 36.3 seconds total.

vooon, 2024-09-14:
What version of the mavlink library you have? Please show rosinstall_generator result file.

XiaoShan262, 2024-10-25:
Hello, I'm having the same problem, has your issue been resolved?

closed 2026-10-11 by github-actions[bot]