mavlink/mavros #1987: ROS2 Quadcopter velocity control / RC Override / Topics and Services Documentation

Sadufy2, 2024-09-13:
### Issue Details

My goal is to write a ROS2 node that can command an IRL drone to move forward and backwards using:
- CubeOrangePlus as the FCU
- Raspberry Pi 4B as an onboard computer.

I'm opening this issue in the hopes of providing some info for anyone coming across the same problems as me and asking for any tips on unsolved errors I have encountered.

----

Using the Ardupilot documentation here (https://ardupilot.org/copter/docs/common-serial-options.html) I was able to,
- Connect the Cube and Pi via USB at Serial0 so it can support MAVLink2.
- SERIAL0_PROTOCOL parameter was set to 2, also to enable MAVLink2.

After cloning the MAVROS source code to "ros2_ws/src" folder, I built the packages (it took around 50min). 
Launch file:

```
# launch/mavros_launch.py

from launch import LaunchDescription
from launch_ros.actions import Node

def generate_launch_description():
    return LaunchDescription([
        Node(
            package='mavros',
            executable='mavros_node',
            output='screen',
            parameters=[{
                'fcu_url': 'serial:///dev/ttyACM0:115200', # ardupilot SERIAL0_BAUD = 115
                'gcs_url': '',
                'target_system_id': 1,
                'target_component_id': 1,
            }],
        ),
    ])
```

The connection was made successfully and all the topics were listed, however at first, all of them were "silent," there wasn't anything published on them except the "/mavros/state" and the "/diagnostics". 
This was fixed by setting the SR0_XXX parameters to 10Hz on the Cube. After that, all sensor data was streaming as expected.

----

After reading this issue #1718 I managed to make some things work (Thanks for the users for sharing their findings):
- I can change Flight Modes using this command: 
```ros2 service call /mavros/set_mode mavros_msgs/srv/SetMode "{base_mode: 0, custom_mode: 'GUIDED_NOGPS'}"```
- And I can also ARM and DISARM the drone using this command: 
```ros2 service call /mavros/cmd/arming mavros_msgs/srv/CommandBool "{value: True}"```
- When I try to send the TAKEOFF command using: 
```ros2 service call /mavros/cmd/takeoff mavros_msgs/srv/CommandTOL "{min_pitch: 0, yaw: 0, altitude: 30}" ```
MavRos console returns success with (result: 0), however, nothing happens IRL, but when the altitude value is increased to 40, it starts returning (result: 4), which is a "Command Denied" error:

```
rosservice call /mavros/cmd/takeoff mavros_msgs/srv/CommandTOL "{min_pitch: 0, yaw: 0, altitude: 40}" 
waiting for service to become available... 
requester: making request: mavros_msgs.srv.CommandTOL_Request(min_pitch=0.0, yaw=0.0, latitude=0.0, longitude=0.0, altitude=40.0)

response: 
mavros_msgs.srv.CommandTOL_Response(success=False, result=4)
```

----

What I gathered so far is:
- I have a good connection to the drone.
- I can read data from the drone.
- I can send some commands to the drone.

----
[... 426 lines cut ...]
  values:
  - key: Voltage
    value: '49.94'
  - key: Current
    value: '-0.4'
  - key: Remaining
    value: '21.0'
- level: "\0"
  name: 'mavros: Heartbeat'
  message: Normal
  hardware_id: uas:///uas1
  values:
  - key: Heartbeats since startup
    value: '44'
  - key: Frequency (Hz)
    value: '1.000051'
  - key: Vehicle type
    value: Quadrotor
  - key: Autopilot type
    value: ArduPilot
  - key: Mode
    value: GUIDED_NOGPS
  - key: System status
    value: STANDBY
```

abz-Soma, 2025-05-30:
yes
im hosting a 'gcs_url' endpoint from the mavros launchfile

```
from launch import LaunchDescription
from launch_ros.actions import Node

def generate_launch_description():
    return LaunchDescription([
        Node(
            package='mavros',
            executable='mavros_node',
            #output='screen',
            parameters=[{
                'fcu_url': 'serial:///dev/ttyACM0:115200',
                'gcs_url': 'udp://@localhost',
                'target_system_id': 1,
                'target_component_id': 1
            }],
        ),
    ])
```

this way you can create a connection to your FCU using pymavlink from any node

```
    def start_mavlink_connection(self):
        self.master = mavutil.mavlink_connection('udp:127.0.0.1:14550')
        self.get_logger().warn("MavLink Connecting... Waiting for heartbeat")
        self.master.wait_heartbeat()
        self.get_logger().info("MavLink Heartbeat - Received")
```

then you can send RC override commands directly to your FCU, without having to use mavros topics

```
from mavros_msgs.msg import RCIn, OverrideRCIn

    def rcOverride_callback(self, msg: OverrideRCIn):
        rc_values = [0] * 8

        for i in range(len(rc_values)):
            rc_values[i] = int(msg.channels[i])

        self.master.mav.rc_channels_override_send(
            self.target_system,
            self.target_component,
            *rc_values
        )
```

if you want to control a channel, give it a valid PWM value, for me channel 4 is back and forward so 

`rc_values[3] = 1600 ` will move the drone forward

**REALLY IMPORTANT: the drone will perform this action until you send a different command. This way your forward/backward stick will be DISABLED and you wont be able to move the drone with your RC. You have to send `rc_values[3] = 0 (invalid pwm)` to release the override and gain back the RC control.**

i suggest having a dedicated button on your RC to switch between manual and auto flight

when flying auto you can send RC_override `[0, 0, 0, 1600, 0, 0, 0, 0]` to move the drone
and when switching to manual always make sure to send  `[0, 0, 0, 0, 0, 0, 0, 0]` to reset the system, because trust me its really fucking scary when your drone starts acting on its own, without the ability to regain control 

also probably use loiter mode while doing this

Sadufy2, 2025-05-30:
yes
im hosting a 'gcs_url' endpoint from the mavros launchfile

```
from launch import LaunchDescription
from launch_ros.actions import Node

def generate_launch_description():
    return LaunchDescription([
        Node(
            package='mavros',
            executable='mavros_node',
            #output='screen',
            parameters=[{
                'fcu_url': 'serial:///dev/ttyACM0:115200',
                'gcs_url': 'udp://@localhost',
                'target_system_id': 1,
                'target_component_id': 1
            }],
        ),
    ])
```

this way you can create a connection to your FCU using pymavlink from any node

```
    def start_mavlink_connection(self):
        self.master = mavutil.mavlink_connection('udp:127.0.0.1:14550')
        self.get_logger().warn("MavLink Connecting... Waiting for heartbeat")
        self.master.wait_heartbeat()
        self.get_logger().info("MavLink Heartbeat - Received")
```

then you can send RC override commands directly to your FCU, without having to use mavros topics

```
from mavros_msgs.msg import RCIn, OverrideRCIn

    def rcOverride_callback(self, msg: OverrideRCIn):
        rc_values = [0] * 8

        for i in range(len(rc_values)):
            rc_values[i] = int(msg.channels[i])

        self.master.mav.rc_channels_override_send(
            self.target_system,
            self.target_component,
            *rc_values
        )
```

if you want to control a channel, give it a valid PWM value, for me channel 4 is back and forward so 

`rc_values[3] = 1600 ` will move the drone forward

**REALLY IMPORTANT: the drone will perform this action until you send a different command. This way your forward/backward stick will be DISABLED and you wont be able to move the drone with your RC. You have to send `rc_values[3] = 0 (invalid pwm)` to release the override and gain back the RC control.**

i suggest having a dedicated button on your RC to switch between manual and auto flight

when flying auto you can send RC_override `[0, 0, 0, 1600, 0, 0, 0, 0]` to move the drone
and when switching to manual always make sure to send  `[0, 0, 0, 0, 0, 0, 0, 0]` to reset the system, because trust me its really fucking scary when your drone starts acting on its own, without the ability to regain control 

also probably use loiter mode while doing this

Sep2000-san, 2025-05-30:
Thank you so much. I will try to follow this advices. 

Now I'm using BlueROV2 control system for a tracked vehicle with 4 motors, and I'm connected using launch file through FCU url. 

I'm able to anything, arming, disarming and change set mode, but when I try to send override command with the topic /mavros/rc/override I'm not able to control each motor separately. 

Instead change 1 of the 18 channel, the number 5, all the motor will move in the same angular direction. 

Instead I'm wondering how I can control each motor.

I'm connected via Ethernet, with Ardupilot firmware, and I have the package Mavros2 on my PC and the Raspberry Pi with Navigator hat.

closed 2026-10-11 by github-actions[bot]