mavlink/mavros #1989: Post-Jazzy and Pre-Jazzy builds have diverged

evan-palmer, 2024-09-14:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details

I was reviewing the CI for #1988, and saw that the Jazzy/Rolling builds succeed, but the addition causes build failures in releases prior to Jazzy (e.g., [humble](https://github.com/mavlink/mavros/actions/runs/10863743005/job/30148243241)).

evan-palmer, 2024-09-14:
It seems like the `ros2` branch may need to split to account for the different dependencies. What do you think @vooon? Do you have ideas for an alternative solution?

vooon, 2024-09-16:
Probably better to add conditionals into CMakeLists.txt. Simply check version of rclcpp and then add or skip libyaml.

evan-palmer, 2024-09-16:
Good idea. I'll submit a PR with that fix.

PR #1994 (2024-09-22): Resolve build error when linking yaml-cpp
This PR addresses the pre-Jazzy build error introduced in #1988, where `yaml-cpp` cannot be found. I've done some testing to verify that things build and run as expected using ROS 2 Humble, Iron, Jazzy, and Rolling.

- Fixes #1989
closed 2024-09-22 by vooon