mavlink/mavros #1995: Upgrade MAVLink in dependencies.rosinstall

evan-palmer, 2024-09-24:
### Issue details

The version of MAVLink targeted in the `dependencies.rosinstall` file (i.e., `release/foxy/mavlink`) does not include the latest changes to pymavlink needed to support Python 3.12 (available in [this](https://github.com/ArduPilot/pymavlink/blob/533161c47995ba47f5548636eed37a47e91072c0/generator/mavgen.py) commit from Tridge). These changes are needed to help address #1958.

evan-palmer, 2024-09-24:
@vooon is there anything I can do to help with this? I took a look at the [mavlink-gpb-release](https://github.com/mavlink/mavlink-gbp-release) repo, and it looks like the bloom build needs to be manually triggered?

vooon, 2024-09-24:
https://github.com/mavlink/mavlink-gbp-release/blob/release/noetic/mavlink/pymavlink/generator/mavgen.py

So 2024.6.6 should have it.

vooon, 2024-09-24:
NOTE: this is the correct one for ROS2: https://github.com/ros2-gbp/mavlink-gbp-release

Foxy is EOLed, and no release accepted to EOLed distros.

evan-palmer, 2024-09-24:
Oh perfect! I hadn't seen that repo. I just did some testing, and things work as expected with Jazzy. Thanks!

closed 2024-09-24 by evan-palmer