mavlink/mavros #1997: No input from RC chanells (mavros/rc/in)

r8d8, 2024-09-30:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details

MAvros doesn't post any rc channel values in /mavros/rc/in topic
Ardupilot's Mavlink Inspectors show that data is present/

### MAVROS version and platform

Mavros: ?0.18.4?
ROS: Humble
Ubuntu: 22.04

### Autopilot type and version

[*] ArduPilot

Version: 4.6.0-dev
Plane

### Node logs

```
ros2 launch mavros apm.launch fcu_url:=udp://@192.168.1.89:14551 gcs_url:=udp://@192.168.191.18:14550                                                                                                                [INFO] [launch]: All log files can be found below /home/juser/.ros/log/2024-09-30-06-12-38-044866-ubuntu-200                                                                                                                                    [INFO] [launch]: Default logging verbosity is set to INFO                                                                                                                                                                                       [INFO] [mavros_node-1]: process started with pid [201]                                                                                                                                                                                          [mavros_node-1] [INFO] [1727676758.721807988] [mavros.mavros_node]: Starting mavros_node container                                                                                                                                              [mavros_node-1] [INFO] [1727676758.722092437] [mavros.mavros_node]: FCU URL: udp://@192.168.1.89:14551                                                                                                                                          [mavros_node-1] [INFO] [1727676758.722162037] [mavros.mavros_node]: GCS URL: udp://@192.168.191.18:14550                                                                                                                                        [mavros_node-1] [INFO] [1727676758.722199893] [mavros.mavros_node]: UAS Prefix: /uas1                                                                                                                                                           [mavros_node-1] [INFO] [1727676758.722235669] [mavros.mavros_node]: Starting mavros router node                                                                                                                                                 [mavros_node-1] [INFO] [1727676758.744738407] [mavros.mavros_router]: Built-in SIMD instructions: ARM NEON                                                                                                                                      [mavros_node-1] [INFO] [1727676758.744867271] [mavros.mavros_router]: Built-in MAVLink package version: 2024.6.6                                                                                                                                [mavros_node-1] [INFO] [1727676758.744916423] [mavros.mavros_router]: Known MAVLink dialects: common ardupilotmega ASLUAV AVSSUAS all csAirLink cubepilot development icarous matrixpilot paparazzi standard storm32 uAvionix ualberta          [mavros_node-1] [INFO] [1727676758.744956999] [mavros.mavros_router]: MAVROS Router started                                                                                                                                                     [mavros_node-1] [INFO] [1727676758.745084040] [mavros.mavros_router]: Requested to add endpoint: type: 0, url: udp://@192.168.1.89:14551                                                                                                        [mavros_node-1] [INFO] [1727676758.745203464] [mavros.mavros_router]: Endpoint link[1000] created                                                                                                                                               [mavros_node-1] [INFO] [1727676758.746718254] [mavros.mavros_router]: link[1000] opened successfully                                                                                                                                            [mavros_node-1] [INFO] [1727676758.746956591] [mavros.mavros_router]: Requested to add endpoint: type: 1, url: udp://@192.168.191.18:14550                                                                                                      [mavros_node-1] [INFO] [1727676758.747062319] [mavros.mavros_router]: Endpoint link[1001] created                                                                                                                                               [mavros_node-1] [INFO] [1727676758.749019702] [mavros.mavros_router]: link[1001] opened successfully                                                                                                                                            [mavros_node-1] [INFO] [1727676758.749238871] [mavros.mavros_router]: Requested to add endpoint: type: 2, url: /uas1                                                                                                                            [mavros_node-1] [INFO] [1727676758.749358039] [mavros.mavros_router]: Endpoint link[1002] created                                                                                                                                               [mavros_node-1] [INFO] [1727676758.755431309] [mavros.mavros_router]: link[1002] opened successfully                                                                                                                                            [mavros_node-1] [INFO] [1727676758.755642222] [mavros.mavros_node]: Starting mavros uas node                                                                                                                                                    [mavros_node-1] [INFO] [1727676758.781061610] [mavros.mavros_router]: link[1001] detected remote address 1.1                                                                                                                                    [mavros_node-1] [INFO] [1727676758.821197531] [mavros.mavros_router]: link[1001] detected remote address 1.154                                                                                                                                  [mavros_node-1] [INFO] [1727676758.885457443] [mavros.mavros]: Plugin actuator_control ignored                                                                                                                                                  [mavros_node-1] [INFO] [1727676758.885491779] [mavros.mavros]: UAS Executor started, threads: 6                                                                                                                                                 [mavros_node-1] [INFO] [1727676758.934131923] [mavros.mavros]: Plugin adsb created                                                                                                                                                              [mavros_node-1] [INFO] [1727676758.934318260] [mavros.mavros]: Plugin adsb initialized                                                                                                                                                          [mavros_node-1] [INFO] [1727676758.934376692] [mavros.mavros]: Plugin altitude ignored                                                                                                                                                          [mavros_node-1] [INFO] [1727676758.954629373] [mavros.mavros]: Plugin cam_imu_sync created                                                                                                                                                      [mavros_node-1] [INFO] [1727676758.954826302] [mavros.mavros]: Plugin cam_imu_sync initialized                                                                                                                                                  [mavros_node-1] [INFO] [1727676758.973720514] [mavros.mavros]: Plugin camera created                                                                                                                                                            [mavros_node-1] [INFO] [1727676758.973923171] [mavros.mavros]: Plugin camera initialized                                                                                                                 [... cut ...]
[mavros_node-1] [WARN] [1727676764.844662629] [mavros.guided_target]: PositionTargetGlobal failed because no origin
[mavros_node-1] [WARN] [1727676764.882513678] [mavros.time]: TM: RTT too high for timesync: 12.93 ms.
[mavros_node-1] [WARN] [1727676765.185157587] [mavros.guided_target]: PositionTargetGlobal failed because no origin
[mavros_node-1] [WARN] [1727676765.481190209] [mavros.time]: TM: RTT too high for timesync: 11.66 ms.
[mavros_node-1] [WARN] [1727676765.523423609] [mavros.guided_target]: PositionTargetGlobal failed because no origin                              
```

[image]

vooon, 2024-09-30:
Please add diagnostics and checkid.

r8d8, 2024-09-30:
Diagnostics fails for me:
```
Traceback (most recent call last):                                                                                                                                                                                                                                File "/usr/bin/rostopic", line 34, in <module>                                                                                                                                                                                                                    import rostopic                                                                                                                                                                                                                                               File "/usr/lib/python3/dist-packages/rostopic/__init__.py", line 65, in <module>                                                                                                                                                                                  import rospy                                                                                                                                                                                                                                                  File "/usr/lib/python3/dist-packages/rospy/__init__.py", line 49, in <module>                                                                                                                                                                                     from .client import spin, myargv, init_node, \                                                                                                                                                                                                                File "/usr/lib/python3/dist-packages/rospy/client.py", line 54, in <module>                                                                                                                                                                                       import rospy.core                                                                                                                                                                                                                                             File "/usr/lib/python3/dist-packages/rospy/core.py", line 75, in <module>                                                                                                                                                                                         from rosgraph_msgs.msg import Log                                                                                                                                                                                                                           ImportError: cannot import name 'Log' from 'rosgraph_msgs.msg' (/opt/ros/humble/local/lib/python3.10/dist-packages/rosgraph_msgs/msg/__init__.py)      
```

vooon, 2024-09-30:
Something must be wrong in your ROS install. Basic ros commands must work.

r8d8, 2024-09-30:
Figured it out,  in issue template there is just old command for ROS1
ROS2 : `ros2 topic echo /diagnostics`
Attached file with diagnostics
[Ros2_diagnostics.txt](https://github.com/user-attachments/files/17186876/Ros2_diagnostics.txt)

r8d8, 2024-09-30:
Checked traffic with Wireshark, 
RC channel message appears, with valid data.
It's should be something on Mavros side.
[image]

r8d8, 2024-09-30:
Found problem.
There is line in rc_io plugin 
```
raw_rc_in.resize(channels.chancount);
```
but MissionPLanner sends `chancount` as zero even if there is data on RC channels.
Checked in debugger. 
Can count it as closed issue, not even Mavros issue. 
Will patch it locally, and think how to deal with it.

closed 2024-09-30 by r8d8