mavlink/mavros #1999: I can't see the mavros topic on wifi

yuan-0816, 2024-10-09:
### Issue details

I am using a Raspberry Pi 4 connected to an Ardupilot Pixhawk 6C, and I use MAVROS to start it. 
When I run ros2 topic list on the Raspberry Pi, I can see the topics being published. 
However, when I run ros2 topic list on another computer (OS is ubuntu 22.04) on the same WiFi network, I can't see the topics.

### MAVROS version and platform

Mavros: 2.8.0
ROS: Humble
Ubuntu: 22.04

### Autopilot type and version

[ x ] ArduPilot
[  ] PX4

Version: ArduCopter V4.5.3

### Node logs

```
drone@drone-desktop:~/Desktop/drone_ws$ ros2 launch flight_control flight_prepare.launch.py 
[INFO] [launch]: All log files can be found below /home/drone/.ros/log/2024-10-09-16-30-42-230050-drone-desktop-8280
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [mavros_node-1]: process started with pid [8290]
[mavros_node-1] [INFO] [1728462643.051490999] [mavros.mavros_node]: Starting mavros_node container
[mavros_node-1] [INFO] [1728462643.051785084] [mavros.mavros_node]: FCU URL: /dev/ttyACM0
[mavros_node-1] [INFO] [1728462643.051829046] [mavros.mavros_node]: GCS URL: 
[mavros_node-1] [INFO] [1728462643.051860285] [mavros.mavros_node]: UAS Prefix: /uas1
[mavros_node-1] [INFO] [1728462643.051887896] [mavros.mavros_node]: Starting mavros router node
[mavros_node-1] [INFO] [1728462643.074448128] [mavros.mavros_router]: Built-in SIMD instructions: ARM NEON
[mavros_node-1] [INFO] [1728462643.074558347] [mavros.mavros_router]: Built-in MAVLink package version: 2024.6.6
[mavros_node-1] [INFO] [1728462643.074593809] [mavros.mavros_router]: Known MAVLink dialects: common ardupilotmega ASLUAV AVSSUAS all csAirLink cubepilot development icarous matrixpilot paparazzi standard storm32 uAvionix ualberta
[mavros_node-1] [INFO] [1728462643.074623401] [mavros.mavros_router]: MAVROS Router started
[mavros_node-1] [INFO] [1728462643.074733213] [mavros.mavros_router]: Requested to add endpoint: type: 0, url: /dev/ttyACM0
[mavros_node-1] [INFO] [1728462643.074813989] [mavros.mavros_router]: Endpoint link[1000] created
[mavros_node-1] [INFO] [1728462643.078372841] [mavros.mavros_router]: link[1000] opened successfully
[mavros_node-1] [INFO] [1728462643.078715925] [mavros.mavros_router]: Requested to add endpoint: type: 2, url: /uas1
[mavros_node-1] [INFO] [1728462643.078815793] [mavros.mavros_router]: Endpoint link[1001] created
[mavros_node-1] [INFO] [1728462643.081405911] [mavros.mavros_router]: link[1000] detected remote address 1.1
[mavros_node-1] [INFO] [1728462643.085689022] [mavros.mavros_router]: link[1001] opened successfully
[mavros_node-1] [INFO] [1728462643.085948219] [mavros.mavros_node]: Starting mavros uas node
[mavros_node-1] [INFO] [1728462643.224852704] [mavros.mavros]: UAS Executor started, threads: 4
[mavros_node-1] [INFO] [1728462643.224967942] [mavros.mavros]: Plugin actuator_control ignored
[mavros_node-1] [INFO] [1728462643.282649470] [mavros.mavros]: Plugin adsb created
[mavros_node-1] [INFO] [1728462643.282821984] [mavros.mavros]: Plugin adsb initialized
[mavros_node-1] [INFO] [1728462643.282871131] [mavros.mavros]: Plugin altitude ignored
[mavros_node-1] [INFO] [1728462643.302213687] [mavros.mavros]: Plugin cam_imu_sync created
[mavros_node-1] [INFO] [1728462643.302409496] [mavros.mavros]: Plugin cam_imu_sync initialized
[mavros_node-1] [INFO] [1728462643.322778303] [mavros.mavros]: Plugin camera created
[mavros_node-1] [INFO] [1728462643.322942207] [mavros.mavros]: Plugin camera initialized
[mavros_node-1] [INFO] [1728462643.343785242] [mavros.mavros]: Plugin cellular_status created
[mavros_node-1] [INFO] [1728462643.343929238] [mavros.mavros]: Plugin cellular_status initialized
[mavros_node-1] [INFO] [1728462643.405587590] [mavros.mavros]: Plugin command created
[mavros_node-1] [INFO] [1728462643.405755363] [mavros.mavros]: Plugin command initialized
[mavros_node-1] [INFO] [1728462643.429050946] [mavros.mavros]: Plugin companion_process_status created
[mavros_node-1] [INFO] [1728462643.429194572] [mavros.mavros]: Plugin companion_process_status initialized
[mavros_node-1] [INFO] [1728462643.429239701] [mavros.mavros]: Plugin debug_value ignored
[mavros_node-1] [INFO] [1728462643.452826184] [mavros.distance_sensor]: DS: rangefinder_pub: Loading config: id: 0
[mavros_node-1] frame_id: lidar
[mavros_node-1] field_of_view: 0.0
[mavros_node-1] send_tf: false
[mavros_node-1] sensor_position: {x: 0.0, y: 0.0, z: -0.1}
[mavros_node-1] [INFO] [1728462643.456851932] [mavros.distance_sensor]: DS: rangefinder_sub: Loading config: subscriber: true
[mavros_node-1] id: 1
[mavros_node-1] orientation: PITCH_270
[mavros_node-1] [INFO] [1728462643.459316831] [mavros.mavros]: Plugin distance_sensor created
[mavros_node-1] [INFO] [1728462643.459578639] [mavros.mavros]: Plugin distance_sensor initialized
[mavros_node-1] [INFO] [1728462643.487361088] [mavros.mavros]: Plugin esc_status created
[... 316 lines cut ...]
/mavros/tunnel/out
/mavros/vfr_hud
/mavros/vision_pose/pose
/mavros/vision_pose/pose_cov
/mavros/vision_speed/speed_twist
/mavros/vision_speed/speed_twist_cov
/mavros/vision_speed/speed_vector
/mavros/wind_estimation
/move_base_simple/goal
/parameter_events
/rosout
/tf
/tf_static
/uas1/mavlink_sink
/uas1/mavlink_source
```

### But I use the another computer can't see it, ```ros2 topic list```
```
yuan@msi-ubuntu:~$ ros2 topic list
/parameter_events
/rosout
```

I check the ROS_DOMAIN_ID before, the two computer's ROS_DOMAIN_ID=1

ajshank, 2024-10-11:
Hi @yuan-0816, this does not seem to be a mavros-specific problem. It's worth investigating if you can `ros2 topic pub` any topic on one machine and `ros2 topic echo` the same on another.

yuan-0816, 2024-10-14:
Yes, I can publish the MAVROS topics by running ```ros2 launch flight_control flight_prepare.launch.py```. However, I cannot receive the topics on another computer. Interestingly, when I publish topics unrelated to MAVROS, the other computer can receive them over WiFi.

Cecilia2005, 2025-03-04:
try this command
```ros2 service call /car/set_stream_rate mavros_msgs/srv/StreamRate "{stream_id: 0, message_rate: 100, on_off: true}"
```
i solve this problem by it,but i do not konw exactly how it work. as far as i know, this command means open all the date stream, rate is 100hz

yuan-0816, 2025-03-05:
@Cecilia2005 謝了老哥 我等等試試 你最帥了

closed 2026-10-11 by github-actions[bot]