mavlink/mavros #2003: Can mavros 2.9.0 be installed by `sudo apt  install  ros-jazzy-mavros`?

YifeiNie, 2024-10-16:
### Issue details

_Please describe the problem, or desired feature_
It seems that mavros 2.8.0 has some bugs making it impossiable to connect FC and mavros, and these bugs are fixed in mavros 2.9.0 and I want to have a try
When I using apt to install mavros, it is still mavros 2.8.0. While using source code to install, when `sudo colcon build`, I met errors as Node logs shown.
So when can we use apt to install mavros 2.9.0 ?
Masters please help me
### MAVROS version and platform

Mavros: 2.9.0
ROS: Jazzy
Ubuntu: 24.04

### Autopilot type and version

[  ] ArduPilot
[  ] PX4
[*] Betaflight

Version: ?3.7.1?

### Node logs
nyf@nyf-OPi5:/opt/ros/jazzy$ sudo colcon build
[3.687s] WARNING:colcon.colcon_ros.task.ament_python.build:Package 'ament_package' doesn't explicitly install a marker in the package index (colcon-ros currently does it implicitly but that fallback will be removed in the future)
[3.693s] WARNING:colcon.colcon_ros.task.ament_python.build:Package 'ament_package' doesn't explicitly install the 'package.xml' file (colcon-ros currently does it implicitly but that fallback will be removed in the future)
[3.701s] WARNING:colcon.colcon_ros.task.ament_python.build:Package 'ament_lint' doesn't explicitly install a marker in the package index (colcon-ros currently does it implicitly but that fallback will be removed in the future)
[3.702s] WARNING:colcon.colcon_ros.task.ament_python.build:Package 'ament_lint' doesn't explicitly install the 'package.xml' file (colcon-ros currently does it implicitly but that fallback will be removed in the future)
[3.728s] WARNING:colcon.colcon_ros.task.ament_python.build:Package 'osrf_pycommon' doesn't explicitly install a marker in the package index (colcon-ros currently does it implicitly but that fallback will be removed in the future)
[3.729s] WARNING:colcon.colcon_ros.task.ament_python.build:Package 'osrf_pycommon' doesn't explicitly install the 'package.xml' file (colcon-ros currently does it implicitly but that fallback will be removed in the future)
Starting >>> ament_package
Starting >>> ament_lint
Starting >>> gtest_vendor
Starting >>> gtest
Starting >>> osrf_pycommon
Starting >>> mavlink
--- stderr: gtest_vendor
CMake Warning:
  Ignoring extra path from command line:

   "/opt/ros/jazzy/share/gtest_vendor"

CMake Error: The source directory "/opt/ros/jazzy/share/gtest_vendor" does not appear to contain CMakeLists.txt.
Specify --help for usage, or press the help button on the CMake GUI.
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/colcon_core/executor/__init__.py", line 91, in __call__
    rc = await self.task(*args, **kwargs)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/colcon_core/task/__init__.py", line 93, in __call__
    return await task_method(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/colcon_ros/task/cmake/build.py", line 34, in build
    rc = await extension.build(
         ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/colcon_cmake/task/cmake/build.py", line 87, in build
    rc = await self._reconfigure(args, env)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/colcon_cmake/task/cmake/build.py", line 182, in _reconfigure
    buildfile = get_buildfile(cmake_cache)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/colcon_cmake/task/cmake/__init__.py", line 114, in get_buildfile
    if 'Visual Studio' in generator:
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: argument of type 'NoneType' is not iterable
---
Failed   <<< gtest_vendor [0.02s, exited with code 1]
Aborted  <<< ament_package [0.02s]
Aborted  <<< ament_lint [0.02s]
Aborted  <<< osrf_pycommon [0.02s]
Aborted  <<< gtest [0.15s]                                             
Aborted  <<< mavlink [0.21s]                                 

Summary: 0 packages finished [3.87s]
  1 package failed: gtest_vendor
  5 packages aborted: ament_lint ament_package gtest mavlink osrf_pycommon
  2 packages had stderr output: gtest_vendor mavlink
  293 packages not processed
argument of type 'NoneType' is not iterable

```
copy output of mavros_node. Usually console where you run roslaunch
```

### Diagnostics

```
place here result of:
rostopic echo -n1 /diagnostics
```

### Check ID

```
rosrun mavros checkid
```

vooon, 2024-10-17:
You should not build in /opt/ros. See source install, it uses work dir at the user. You anyway will need it to host your extra scripts.

http://repo.ros2.org/status_page/ros_jazzy_default.html?q=mavros - seems 2.9 already out.

vooon, 2024-10-17:
Ah, well, it in testing: https://repo.ros2.org/ubuntu/testing/pool/main/r/ros-jazzy-mavros/

For main - need to wait next sync. https://repo.ros2.org/ubuntu/main/pool/main/r/ros-jazzy-mavros/

closed 2026-10-11 by github-actions[bot]