mavlink/mavros #2012: mavros failed to convert rostopic to mavlink

YifeiNie, 2024-12-19:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details

\for ROS1 noetic, FCU to mavros is of which is shown in `rostopic echo /mavlink/from`, and I can also echo topics like IMU data. THen I published some topics like `mavros/setpoint_raw/attitude` and tried to send command to FCU, this topic can also seen by `rostopic echo` and is correct. But `rostopic echo /mavlink/to` has no data, which means no data from mavros to FCU
What's strange is the same option in ROS2 jazzy, all things are ok

Masters please help me

### MAVROS version and platform

Mavros: ?0.18.4?
ROS: noetic
Ubuntu: 20.04

### Autopilot type and version

[  ] ArduPilot
[  ] PX4

Version: ?3.7.1?

### Node logs

```
copy output of mavros_node. Usually console where you run roslaunch
```

### Diagnostics

```
place here result of:
rostopic echo -n1 /diagnostics
```

### Check ID

```
rosrun mavros checkid
```

vooon, 2024-12-20:
ROS1 and ROS2 has quite different communication layer. In ROS2 there a Router, which expose topics for UAS (to/from), and a UAS (with bunch of the plugins), which talks to the Router.

ROS1 do not have router, and UAS talks directly to MAVConn, so you do not have access to the bus. To/from may only be used by external gcs bridge (from extras), or if you need direct mavlink communication to the FCU.

YifeiNie, 2024-12-20:
> ROS1 and ROS2 has quite different communication layer. In ROS2 there a Router, which expose topics for UAS (to/from), and a UAS (with bunch of the plugins), which talks to the Router.
>
> ROS1 do not [...]

@vooon tkf for quick reply!
As you said, the process mavros (ros1) converts rostopic to mavlink will not shown in `rostopic echo /mavlink/to` although it is working.
But my FCU can not receive data from mavros in ros1, but my FCU can do so in ros2 with the same `px4.launch` file
So could you please tell me the differences between mavros for ros1 and ros2 in mavlink data transmitting. How can I fix this?

YifeiNie, 2024-12-20:
> > ROS1 and ROS2 has quite different communication layer. In ROS2 there a Router, which expose topics for UAS (to/from), and a UAS (with bunch of the plugins), which talks to the Router.
> > ROS1 do not have router, and UAS talks directly to MAVConn, so you do not have access to the bus. To/from may only be used by external gcs bridge (from extras), or if you need direct mavlink communication to the FCU.
> 
> @vooon tks for quick reply! As you said, the process mavros (ros1) converts rostopic to mavlink will not shown in `rostopic echo /mavlink/to` although it is working. But my FCU can not receive data from mavros in ros1, but my FCU can do so in ros2 with the same `px4.launch` file So could you please tell me the differences between mavros for ros1 and ros2 in mavlink data transmitting. How can I fix this? Thank you so much!

This question solved without any changes, maybe I fegot to save my operations and led to this problem.
Anyway, thanks in advance

closed 2025-01-02 by vooon