mavlink/mavros #2014: Serial port baud rates are dependent and restricted by the version of asio

slovak194, 2025-01-02:
### Issue details

Serial port baud rates are dependent and restricted by the version of asio. Particularly, in the serial_port_base.ipp some baud rates might be missing but still supported by the OS with termios/termios2/ioctl. In my case, it is the 1500000 baud rate which throws the Device exceptio within MAVROS but works fine if set manually. In the latest asio master it is still missing.
https://github.com/chriskohlhoff/asio/blob/62481a25be6cf78cbe714419a4462fd89bd84ab9/asio/include/asio/impl/serial_port_base.ipp#L93C8-L101C40

```C++
# ifdef B1000000
  case 1000000: baud = B1000000; break;
# endif
# ifdef B1152000
  case 1152000: baud = B1152000; break;
# endif
# ifdef B2000000
  case 2000000: baud = B2000000; break;
```

A proposed solution might be to split the try-catch in the serial device constructor into sections and try one more time with serial_dev.native_handle() and ioctl if the baud rate option causes an exception.

### MAVROS version and platform

Mavros: N/A
ROS: N/A
Ubuntu: 22.04

Best regards,
Alex

vooon, 2025-01-02:
I'd rather not to reinvent the wheel with that termios (as it were here at early days...).
Also are you sure, your UART works fine on such high rates?

slovak194, 2025-01-03:
>> I'd rather not to reinvent the wheel with that termios (as it were here at early days...).

That is true, but what is the alternative if we have to deal with hardcoded baud rates in ASIO? It worked well for me until now.

>> Also are you sure, your UART works fine on such high rates?

Ardupilot specifies literaly "Most stm32-based boards can support rates of up to 1500". Hardware-wise serial ports on modern controllers might work at 12.5 MBaud/s in PPP mode. On my end 1500000 bps and 2000000 bps work fine even with finicky cables.

vooon, 2025-01-03:
Well, you can just package up fresh version of the ASIO, or make a patched version.

slovak194, 2025-02-24:
> The asio libraries are well optimised, it would be disservice to start from scratch.

I feel there is a misunderstanding. No reason to start anything from scratch. ASIO does not aim to handle every aspect of serial communication, which is why it provides the native_handle() method for such things.

BR,
Alex

closed 2026-10-10 by github-actions[bot]