mavlink/mavros #2020: ROS1 Mavros global_position plugin not publishing fused global position topics unless number of satellites >= 15

DARKMOONlite, 2025-02-21:
In my ros1 & px4 project I've experienced that mavros only publishes the fused global position topics `~/global`, `~/local`, `~/rel_alt` & `~/compass_hdg` if the number of satellites visible is greater or equal to 15, despite correctly interpreting the mavlink data which is also getting published on the `raw/fix` topic. 
Despite being in a relatively large field near a lake, during my flight tests if the weather is clear and I'm lucky I can sometimes get 15-20 visible satellites, however on cloudy days or just 'bad' days in general, I usually hover around 10 satellites with a 1 - 1.5 `hdop`/`vdop`. I'm wondering if this is desired behaviour or if I'm missing something, as I've read through the documentation and the [global_position.cpp file](https://github.com/mavlink/mavros/blob/master/mavros/src/plugins/global_position.cpp), none of which state a required number of satellites beyond that of px4's internal required default `6` satellites in order to take off. Not only does my code expect msgs over the `~/global` topic but also it expects a transform being published between the `map` and `base_link` frames which again is only published to `/tf` when the number of visible satellites >=15. For the time being I'm going to create a new node that reads the incoming msgs from the `~/raw/fix` topic and forward it to the `~/global` topic and `/tf` if the number of visible satellites is under 15 satellites, if that's not a good solution then I am all ears for a better stopgap or permanent fix. 

Below is my `config` file that and `pluginlists` files I'm passing to mavros. Any guidance or help would be great. If i learn any more in the following days/weeks I'll publish here for posterity. 
```yaml
plugin_blacklist: # https://github.com/mavlink/mavros/tree/master/mavros_extras
  - 3dr_radio
  - actuator_control
  - adsb
  - cam_imu_sync
  - cellular_status
  - distance_sensor
  - esc_status
  - esc_telemtery
  - esc_telemetry
  - ftp
  - geofence
  - gps_rtk
  - guided_target
  - image_pub
  - log_transfer
  - mag_calibration_status
  - manual_control
  - mocap_pose_estimate
  - mount_control
  - nav_controller_output
  - obstacle_distance
  - play_tune
  - px4flow
  - rallypoint
  - rangefinder
  - rc_io
  - safety_area
  - terrain
  - tunnel
  - vibration
  - vision_pose_estimate
  - vision_speed_estimate
  - wheel_odometry
  - wind_estimation
  - fake_gps

plugin_whitelist:
- 'sys_*' # this needs to be enabled for mavros to work
- 'setpoint_*' # loads all the setpoint plugins to allow for setting desitinations
- altitude
- camera
- command
- companion_process_status
- debug_value
- gps_input
- gps_status
#- guided_target # removed to solve the `PositionTargetGlobal failed because no origin` error, 
- hil
- home_position
- imu
- local_position
- odom # from mavros_extras https://github.com/mavlink/mavros/blob/master/mavros_extras/src/plugins/odom.cpp
- onboard_computer_status
- param
- trajectory
- waypoint
- heartbeat_status

```

```yaml
# Common configuration for PX4 autopilot
#
# node:
[... 253 lines cut ...]
# vibration
vibration:
  frame_id: "base_link"

# wheel_odometry
wheel_odometry:
  count: 2           # number of wheels to compute odometry
  use_rpm: false     # use wheel's RPM instead of cumulative distance to compute odometry
  wheel0: {x: 0.0, y: -0.15, radius: 0.05} # x-, y-offset (m,NED) and radius (m)
  wheel1: {x: 0.0, y:  0.15, radius: 0.05} # x-, y-offset (m,NED) and radius (m)
  send_raw: true              # send wheel's RPM and cumulative distance (~/wheel_odometry/rpm, ~/wheel_odometry/distance)
  send_twist: false           # send geometry_msgs/TwistWithCovarianceStamped instead of nav_msgs/Odometry
  frame_id: "odom"             # origin frame
  child_frame_id: "base_link" # body-fixed frame
  vel_error: 0.1              # wheel velocity measurement error 1-std (m/s)
  tf:
    send: false
    frame_id: "odom"
    child_frame_id: "base_link"

# camera
camera:
  frame_id: "base_link"

```

vooon, 2025-02-22:
Check your FC config. If the autopilot not sending global position - how can mavros do?

DARKMOONlite, 2025-02-23:
I believe my FC is sending global position data, if this isn't a feature of mavros then I'll look into my flight controller settings to see if I see anything.

closed 2026-10-10 by github-actions[bot]