mavlink/mavros #2030: colcon build Failed While Trying to Install from Source

DianYueGuo, 2025-04-17:
[complete_log_of_installation.log](https://github.com/user-attachments/files/19798908/complete_log_of_installation.log)

I tried to follow the [instruction](https://github.com/mavlink/mavros/blob/ros2/mavros/README.md#source-installation) to install mavros from source, and failed to build `fastrtps`. The installation was run in a docker container preinstalled with ROS2 humble inside a Raspberry PI (Armv7).

After the failed build, I tried to run the `mavros_node`, which of course didn't work:
``` terminal
root@raspberrypi32:~/mavros_ros2_ws# ros2 run mavros mavros_node --ros-args --params-file params.yaml
Package 'mavros' not found
```

The Dockerfile for the docker container I used (image: [dianyueguo/orca-auv-rpi-ros2-image:latest](https://hub.docker.com/r/dianyueguo/orca-auv-rpi-ros2-image/tags)):
``` Dockerfile
FROM ubuntu:22.04

SHELL ["/bin/bash", "-c"]

ENV ROS_DISTRO=humble

RUN locale  # check for UTF-8
RUN apt-get update && apt-get install locales
RUN locale-gen en_US en_US.UTF-8
RUN update-locale LC_ALL=en_US.UTF-8 LANG=en_US.UTF-8
ENV LANG=en_US.UTF-8
RUN locale  # verify settings

RUN apt-get install software-properties-common -y
RUN add-apt-repository universe

RUN apt-get update && apt-get install curl -y
RUN curl -sSL https://raw.githubusercontent.com/ros/rosdistro/master/ros.key -o /usr/share/keyrings/ros-archive-keyring.gpg

RUN echo "deb [arch=$(dpkg --print-architecture) signed-by=/usr/share/keyrings/ros-archive-keyring.gpg] http://packages.ros.org/ros2/ubuntu $(. /etc/os-release && echo $UBUNTU_CODENAME) main" | tee /etc/apt/sources.list.d/ros2.list > /dev/null

# Set environment variable to prevent interactive prompts
ENV DEBIAN_FRONTEND=noninteractive
RUN apt-get update && apt-get install -y \
        python3-flake8-docstrings \
        python3-pip \
        python3-pytest-cov \
        ros-dev-tools
# Reset DEBIAN_FRONTEND to avoid issues with interactive shells later
ENV DEBIAN_FRONTEND=dialog

RUN apt-get install -y \
        python3-flake8-blind-except \
        python3-flake8-builtins \
        python3-flake8-class-newline \
        python3-flake8-comprehensions \
        python3-flake8-deprecated \
        python3-flake8-import-order \
        python3-flake8-quotes \
        python3-pytest-repeat \
        python3-pytest-rerunfailures

RUN mkdir -p ~/ros2_humble/src && \
    cd ~/ros2_humble && \
    vcs import --input https://raw.githubusercontent.com/ros2/ros2/humble/ros2.repos src

RUN sudo apt-get upgrade

RUN rosdep init
RUN rosdep update
RUN rosdep install --from-paths ~/ros2_humble/src --ignore-src -y --skip-keys "fastcdr rti-connext-dds-6.0.1 urdfdom_headers"

RUN cd ~/ros2_humble/ && \
    colcon build --symlink-install

RUN echo "source ~/ros2_humble/install/local_setup.bash" >> /etc/bash.bashrc

RUN apt-get update && apt-get install -y git

RUN source ~/ros2_humble/install/local_setup.bash && \
    cd ~/ && \
    mkdir uros_ws && cd uros_ws && \
    git clone -b humble https://github.com/micro-ROS/micro_ros_setup.git src/micro_ros_setup && \
    rosdep update && rosdep install --from-paths src --ignore-src -y && \
    colcon build && \
    source install/local_setup.bash && \
    ros2 run micro_ros_setup create_agent_ws.sh && \
    ros2 run micro_ros_setup build_agent.sh

RUN echo "source ~/uros_ws/install/local_setup.bash" >> /etc/bash.bashrc

RUN apt-get install gcc build-essential cmake rpm libusb-1.0 libusb-1.0-0-dev libgtk-3-dev pandoc -y

RUN mkdir ~/git
WORKDIR /root/git
RUN git clone https://github.com/stlink-org/stlink.git
WORKDIR /root/git/stlink
RUN make clean
RUN make release
RUN make install
RUN ldconfig
RUN make debug
RUN make package

WORKDIR /root

CMD ["/bin/bash"]
```

The installation command:
``` bash
apt install -y python3-vcstool python3-rosinstall-generator python3-osrf-pycommon

mkdir -p ~/mavros_ros2_ws/src
cd ~/mavros_ros2_ws

rosinstall_generator --format repos mavlink --deps | tee /tmp/mavlink.repos

rosinstall_generator --format repos --upstream mavros --deps | tee -a /tmp/mavros.repos

vcs import src < /tmp/mavlink.repos
vcs import src < /tmp/mavros.repos
rosdep install --from-paths src --ignore-src -y

./src/mavros/mavros/scripts/install_geographiclib_datasets.sh

colcon build

source install/setup.bash
```

The failure in `colcon build`:
``` terminal
--- stderr: fastrtps                                                                                                                                                               
In file included from /root/mavros_ros2_ws/src/fastrtps/include/fastrtps/types/TypeObject.h:18,
                 from /root/mavros_ros2_ws/src/fastrtps/include/fastdds/dds/core/policy/QosPolicies.hpp:36,
                 from /root/mavros_ros2_ws/src/fastrtps/include/fastdds/dds/subscriber/qos/ReaderQos.hpp:23,
                 from /root/mavros_ros2_ws/src/fastrtps/include/fastrtps/qos/ReaderQos.h:23,
                 from /root/mavros_ros2_ws/src/fastrtps/include/fastdds/rtps/participant/RTPSParticipant.h:30,
                 from /root/mavros_ros2_ws/src/fastrtps/src/cpp/rtps/network/NetworkFactory.cpp:22:
/root/mavros_ros2_ws/src/fastrtps/include/fastrtps/types/TypesBase.h:285:27: error: expected unqualified-id before numeric constant
  285 | #define MEMBER_ID_INVALID 0X0FFFFFFF
      |                           ^~~~~~~~~~
In file included from /root/mavros_ros2_ws/src/fastrtps/src/cpp/rtps/transport/tcp/RTCPHeader.h:21,
                 from /root/mavros_ros2_ws/src/fastrtps/src/cpp/rtps/transport/TCPTransportInterface.h:21,
                 from /root/mavros_ros2_ws/src/fastrtps/src/cpp/rtps/network/NetworkFactory.cpp:27:
/root/mavros_ros2_ws/install/fastcdr/include/fastcdr/Cdr.h: In member function ‘eprosima::fastcdr::Cdr& eprosima::fastcdr::Cdr::operator<<(const _T&)’:
/root/mavros_ros2_ws/install/fastcdr/include/fastcdr/Cdr.h:329:31: error: no match for ‘operator==’ (operand types are ‘int’ and ‘eprosima::fastcdr::MemberId’)
  329 |         if (MEMBER_ID_INVALID == next_member_id_)
      |                               ^~ ~~~~~~~~~~~~~~~
      |                                  |
      |                                  eprosima::fastcdr::MemberId
/root/mavros_ros2_ws/install/fastcdr/include/fastcdr/Cdr.h: In member function ‘eprosima::fastcdr::Cdr& eprosima::fastcdr::Cdr::operator>>(_T&)’:
/root/mavros_ros2_ws/install/fastcdr/include/fastcdr/Cdr.h:359:31: error: no match for ‘operator==’ (operand types are ‘int’ and ‘eprosima::fastcdr::MemberId’)
  359 |         if (MEMBER_ID_INVALID == next_member_id_)
      |                               ^~ ~~~~~~~~~~~~~~~
      |                                  |
      |                                  eprosima::fastcdr::MemberId
gmake[2]: *** [src/cpp/CMakeFiles/fastrtps.dir/build.make:678: src/cpp/CMakeFiles/fastrtps.dir/rtps/network/NetworkFactory.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:941: src/cpp/CMakeFiles/fastrtps.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< fastrtps [7min 41s, exited with code 2]
Aborted  <<< rosidl_runtime_c [1min 32s]                                                                                                                               
Aborted  <<< mavlink [1min 55s]                                                                                              
Aborted  <<< iceoryx_posh [7min 9s]                                             

Summary: 81 packages finished [13min 4s]
  1 package failed: fastrtps
  3 packages aborted: iceoryx_posh mavlink rosidl_runtime_c
  15 packages had stderr output: ament_cmake_gen_version_h ament_index_cpp fastcdr fastrtps iceoryx_hoofs iceoryx_posh mavlink mimick_vendor osrf_testing_tools_cpp performance_test_fixture rcutils rosidl_runtime_c rosidl_typesupport_interface rti_connext_dds_cmake_module tracetools
  118 packages not processed
```

Ryanf55, 2025-04-23:
Why are you building fastdds from source in Ubuntu? OSRF distributes binaries.

DianYueGuo, 2025-04-26:
I am supposing that they don't have binary supports for Armv7 platforms, so I have to build it from source.

Ryanf55, 2025-04-26:
Ah for Mac, yes, build from source. Please be aware that MAC is community supported and can be difficult to maintain due the rolling nature of the distro. I don't have one, so sorry, I can't help much.

DianYueGuo, 2025-04-27:
Ryan,

Thank you very much for your reply.

My goal was to install mavros into my docker container that needs to be built on Armv7 platform for Raspberry PI. Some of the ROS2 packages don't have binary supports for that platform, so I thought that all of the dependencies needs to be built from source in order to install mavros. Unfortunately, package `fastrtps` ran into erros while building.

After a while of trial and error, I found a solution for my problem. I only have to install some of the dependencies from source, and not all of them. Some of the dependencies might already have been installed during the installation of ROS2 Humble, and some other dependencies might have binary supports for the platform.

Here is the final Dockerfile I made with the installation of mavros added:
```Dockerfile
FROM ubuntu:22.04

SHELL ["/bin/bash", "-c"]

ENV ROS_DISTRO=humble

RUN locale  # check for UTF-8
RUN apt-get update && apt-get install locales
RUN locale-gen en_US en_US.UTF-8
RUN update-locale LC_ALL=en_US.UTF-8 LANG=en_US.UTF-8
ENV LANG=en_US.UTF-8
RUN locale  # verify settings

RUN apt-get install software-properties-common -y
RUN add-apt-repository universe

RUN apt-get update && apt-get install curl -y
RUN curl -sSL https://raw.githubusercontent.com/ros/rosdistro/master/ros.key -o /usr/share/keyrings/ros-archive-keyring.gpg

RUN echo "deb [arch=$(dpkg --print-architecture) signed-by=/usr/share/keyrings/ros-archive-keyring.gpg] http://packages.ros.org/ros2/ubuntu $(. /etc/os-release && echo $UBUNTU_CODENAME) main" | tee /etc/apt/sources.list.d/ros2.list > /dev/null

# Set environment variable to prevent interactive prompts
ENV DEBIAN_FRONTEND=noninteractive
RUN apt-get update && apt-get install -y \
        python3-flake8-docstrings \
        python3-pip \
        python3-pytest-cov \
        ros-dev-tools
# Reset DEBIAN_FRONTEND to avoid issues with interactive shells later
ENV DEBIAN_FRONTEND=dialog

RUN apt-get install -y \
        python3-flake8-blind-except \
        python3-flake8-builtins \
        python3-flake8-class-newline \
        python3-flake8-comprehensions \
        python3-flake8-deprecated \
        python3-flake8-import-order \
        python3-flake8-quotes \
        python3-pytest-repeat \
        python3-pytest-rerunfailures

RUN mkdir -p ~/ros2_humble/src && \
    cd ~/ros2_humble && \
    vcs import --input https://raw.githubusercontent.com/ros2/ros2/humble/ros2.repos src

RUN sudo apt-get upgrade

RUN rosdep init
RUN rosdep update
RUN rosdep install --from-paths ~/ros2_humble/src --ignore-src -y --skip-keys "fastcdr rti-connext-dds-6.0.1 urdfdom_headers"

RUN cd ~/ros2_humble/ && \
    colcon build --symlink-install

RUN echo "source ~/ros2_humble/install/local_setup.bash" >> /etc/bash.bashrc

RUN apt-get update && apt-get install -y git

RUN source ~/ros2_humble/install/local_setup.bash && \
    cd ~/ && \
    mkdir uros_ws && cd uros_ws && \
    git clone -b humble https://github.com/micro-ROS/micro_ros_setup.git src/micro_ros_setup && \
    rosdep update && rosdep install --from-paths src --ignore-src -y && \
    colcon build && \
    source install/local_setup.bash && \
    ros2 run micro_ros_setup create_agent_ws.sh && \
    ros2 run micro_ros_setup build_agent.sh

RUN echo "source ~/uros_ws/install/local_setup.bash" >> /etc/bash.bashrc

RUN apt-get install gcc build-essential cmake rpm libusb-1.0 libusb-1.0-0-dev libgtk-3-dev pandoc -y

RUN mkdir ~/git
WORKDIR /root/git
RUN git clone https://github.com/stlink-org/stlink.git
WORKDIR /root/git/stlink
RUN make clean
RUN make release
RUN make install
RUN ldconfig
RUN make debug
RUN make package

RUN apt install python3-rosinstall-generator

RUN mkdir -p /root/mavros_ws/src
WORKDIR /root/mavros_ws

RUN rosinstall_generator --format repos mavlink | tee /tmp/mavlink.repos
RUN rosinstall_generator --format repos --upstream mavros | tee -a /tmp/mavros.repos

RUN vcs import src < /tmp/mavlink.repos
RUN vcs import src < /tmp/mavros.repos

WORKDIR /root/mavros_ws/src
RUN git clone https://github.com/ros/diagnostics.git -b 4.0.3
RUN git clone https://github.com/ros-geographic-info/geographic_info.git -b 1.0.6
RUN git clone https://github.com/ros/eigen_stl_containers.git -b 1.1.0
RUN git clone https://github.com/ros/angles.git -b 1.15.0

WORKDIR /root/mavros_ws/
RUN apt-get update
RUN source ~/ros2_humble/install/setup.bash && rosdep install --from-paths src --ignore-src -y

RUN ./src/mavros/mavros/scripts/install_geographiclib_datasets.sh

RUN source ~/ros2_humble/install/setup.bash && colcon build

RUN echo "source ~/mavros_ws/install/setup.bash" >> /etc/bash.bashrc

WORKDIR /root

CMD ["/bin/bash"]
```

DianYueGuo, 2025-04-27:
Although the build failure issue isn't solved, my problem is. So, I suppose the issue should be closed.

closed 2025-04-27 by DianYueGuo