mavlink/mavros #2031: CPU usage 4x higher in ROS 2 than ROS 1

Ryanf55, 2025-04-22:
Hello,

When testing the same MAVROS set of plugins and stream rates, we observe MAVROS using 4x the CPU usage of ROS1.
Is this expected, or are there any mitigations?

Best,
Ryan

vooon, 2025-04-25:
Well, ROS2 version is by design have more nodes as of each plugin is a node.
DDS also can have more consumption, don't know. Ideally i'd like to see something OTEL for ROS2.

Ryanf55, 2025-04-25:
I'll do my best to find a way to reproduce this in SITL, outside of our hardware. We suspect that changing executors may help.

Ryanf55, 2025-09-22:
The events executor in ROS 2 jazzy helps a lot, but seemed to cause service failures only in mavros.

jacog2, 2026-05-11:
Submitted PR #2189 as a low-risk mitigation path for the ROS 2 CPU overhead reported here: https://github.com/mavlink/mavros/pull/2189

It adds a documented `MAVROS_UAS_EXECUTOR_THREADS` override so deployments can run the UAS executor with 2 worker threads without forcing an EventsExecutor switch for all users.

shupx, 2026-10-10:
https://blog.csdn.net/benchuspx/article/details/167495626