mavlink/mavros #2041: Loopback of DISTANCE_SENSOR messages

schashni, 2025-05-14:
**Setup**
I'm sending rangefinder data to the PixHawk (SITL) via MAVROS.

Configuration of the distance_sensor:
```
mavros/distance_sensor:
  ros__parameters:
    config: |
      altimeter_sub:
        subscriber: true
        id: 0
        orientation: PITCH_270 
```
**What happens**
1) The PixHawk receives MAVLink messages and processes them. I can see this in the QGC.
2) Then, the PixHawk sends them back via MAVLink.
3) MAVROS receives these messages and prints an error:

> [mavros_node-7] [ERROR] [1747261264.493635734] [mavros.distance_sensor]: DS: altimeter_sub (id 0) is subscriber, but i got sensor data for that id from FCU

4) In the end, my task (transmitting rangefinder data to the PixHawk) is solved, but there is a constant flow of error messages in the terminal.

**MAVROS version and platform**
Mavros: ros2 HEAD (https://github.com/mavlink/mavros/commit/8cfe42e0f31d53899fab0b15bdd7e5f758eab067)
ROS: Humble
Ubuntu: 22.04
Autopilot type and version

[ x ] ArduPilot
[ ] PX4

Version: ArduSub V4.6.0-dev (0805156d)

Running ArduSub in SITL.

**Comments**
I'm new to MAVROS, so maybe I'm missing some concern.
From my point of view MAVROS should ignore these messages, since their directions are set in the configuration file. Or maybe I'm unaware of some parameter in the configuration file.

vooon, 2025-05-15:
It was expected that PX4 not to route message back, looks like something have changed over time.
And the error message was to help to find out possible misconfiguration.

ZephyrKeXiner, 2025-05-23:
I encountered the similar problem.While the mavros starts,the log says 
"[INFO] [1748004972.939274818] [mavros.distance_sensor]: DS: Plugin not configured!
[INFO] [1748004972.939313985] [mavros]: Plugin distance_sensor created
[INFO] [1748004972.939322443] [mavros]: Plugin distance_sensor initialized"

My controller is CUAV-X7 (Ardupilot).
I have checked the file "apm_pluginlist.yaml" and I guarantee that the plugin "distance_sensor" isn't on the blacklist.
What should I check next and how to fix the bug

closed 2026-10-09 by github-actions[bot]