mavlink/mavros #2068: ROS2: Documentation book

vooon, 2025-09-17:
Ros wiki is outdated and not really helpful for a new people to find info.
Need to make something to generate plugin documentation.

peakyquest, 2025-10-18:
I'm considering whether it would be appropriate to open a pull request to contribute MAVROS-related examples, specifically focusing on services and topics. There are two potential approaches: one is to start with an initial set of essential examples by creating a mavros_examples package and then encourage community contributions by opening “good first issues” for additional examples. The other is to wait until a complete and comprehensive set of examples is ready before merging. The purpose of the examples would be to address common pain points—particularly misunderstandings around how specific MAVROS topics and services behave or are expected to function. These examples would serve as clear reference implementations to help users avoid frequent mistakes. I’d appreciate feedback on which direction would be more suitable for the project, and I’m happy to proceed with either.

vooon, 2025-10-18:
I think `mavros_examples` would be good, similar to turtlebot or other study bots, where you can run some common examples.